natevw/node-nrf #31: add startListening and stopListening methods

atomicus, 2015-01-28:
As nrf is half-duplex, it would be nice to have 2 methods on rx pipe.
startListening
stopListening

just to 'pause' when we want to tx something, without a need closing of re-initializing of rx pipe.
This would also make your problem of pipe adresse shortage less important.

natevw, 2015-01-30:
The idea is that node-nrf already manages this for you. You may open as many TX pipes as you'd like, and when you transmit it automatically pauses RX as necessary. Please let me know if this is not working for you.

closed 2015-01-30 by natevw

atomicus, 2015-01-30:
Ok, I thought that is not happening as I got a lot of packet loss during transmission, but that could be only random issue.

natevw, 2015-01-30:
You may get a certain amount of packet loss, anything that a remote transmitter sends while you are not in RX mode will be lost. So if your channel is very busy that could be a lot of packets.

But if it is more than you would expect, there is a chance be that something is not switching it back as quickly as it should. If you see any clues to that with `nrf._debug = true` on, I'm happy to reopen this issue and investigate further.