natevw/node-nrf #35: nRF24LU1 backend

natevw, 2015-03-01:
Nordic also offer the nRF24LU1, a SoC that I'd like to develop a basic "passthrough" firmware — then instead of having to wire up an nRF24 breakout board via the GPIO pins, you could simply plug one into USB (even on e.g. a MacBook!) and be done…

Basic ideas:

- probably make the SoC enumerate as an HID, 64 byte packet limit there should be more than sufficient and then no drivers necessary
- probably need to invert the xcvr_base vs. xcvr_api class hierarchy, so that we can override at the getStates/setStates/execCommand level [this is probably a better way to support all alternate backends anyway…]

c.f. the [Geeetech Crazyradio](http://www.amazon.com/Geeetech-Crazyradio-Crazyflie-Quadcopter-Helicopter/dp/B00GNXVLR4) (built primarily to control a little quadcopter) which has a nice external antenna.

Manawyrm, 2015-03-07:
brillant idea. You can buy LU1 dongles in china for under 10$. 
Programming them on the other side is unfortunatly a bit strange/hard, you either need to buy the really expensive (500$!!) genuine programmer, or build one of the hacky-versions (with LPC-arm-core)...

natevw, 2015-03-18:
Another note on this: for HID the throughput will be limited, which for many use cases will be fine, but no where near the 2Mbps RF link rate. An alternate (or perhaps primary, based on driver situation…) backend based on the "default" firmware with separate control/bulk endpoints might be more general. If I can communicate with the USB device without an OS-level driver, this would be the way to go.