natevw/node-nrf #36: Error when installing library in raspberry pi

jlurgo, 2015-03-03:
Hi! 
I´ve been failing to install the library in raspberry pi. 
Can you tell me please what I´m doing wrong?
I flashed the Pi with the latest Raspbian image, and upgraded it with:
sudo apt-get update
sudo apt-get upgrade

Then I installed node with:
wget http://node-arm.herokuapp.com/node_latest_armhf.deb
sudo dpkg -i node_latest_armhf.deb

Then in my node_project folder:
npm install nrf

And it throws me the following errors:

> pi-spi@0.8.7 install /home/pi/apps/PruebaNrf/node_modules/nrf/node_modules/pi-spi
> node-gyp rebuild

child_process: customFds option is deprecated, use stdio instead.
make: Entering directory '/home/pi/apps/PruebaNrf/node_modules/nrf/node_modules/pi-spi/build'
  CXX(target) Release/obj.target/spi_binding/spi_binding.o
../spi_binding.cc:18:1: error: ‘uv_mutex_t’ does not name a type
../spi_binding.cc:23:5: error: ‘uv_work_t’ does not name a type
../spi_binding.cc:36:16: error: variable or field ‘_Transfer’ declared void
../spi_binding.cc:36:16: error: ‘uv_work_t’ was not declared in this scope
../spi_binding.cc:36:27: error: ‘req’ was not declared in this scope
spi_binding.target.mk:81: recipe for target 'Release/obj.target/spi_binding/spi_binding.o' failed
make: *** [Release/obj.target/spi_binding/spi_binding.o] Error 1
make: Leaving directory '/home/pi/apps/PruebaNrf/node_modules/nrf/node_modules/pi-spi/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1067:12)
gyp ERR! System Linux 3.18.7+
gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/apps/PruebaNrf/node_modules/nrf/node_modules/pi-spi
gyp ERR! node -v v0.12.0
gyp ERR! node-gyp -v v1.0.2
gyp ERR! not ok

> epoll@0.1.12 install /home/pi/apps/PruebaNrf/node_modules/nrf/node_modules/pi-pins/node_modules/epoll
> node-gyp rebuild

child_process: customFds option is deprecated, use stdio instead.
make: Entering directory '/home/pi/apps/PruebaNrf/node_modules/nrf/node_modules/pi-pins/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  SOLINK_MODULE(target) Release/obj.target/epoll.node: Finished
  COPY Release/epoll.node
make: Leaving directory '/home/pi/apps/PruebaNrf/node_modules/nrf/node_modules/pi-pins/node_modules/epoll/build'
npm ERR! Linux 3.18.7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "nrf"
npm ERR! node v0.12.0
npm ERR! npm  v2.5.1
npm ERR! code ELIFECYCLE

npm ERR! pi-spi@0.8.7 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the pi-spi@0.8.7 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the pi-spi package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls pi-spi
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/apps/PruebaNrf/npm-debug.log

Can you tell me how can I get through this?
Thanks in advance, excuse me if this is too basic, I´m new to linux.

natevw, 2015-03-03:
I've updated a dependency (pi-spi) for node 0.12.x but node-nrf itself is still requesting the older version. You can try either:

- `npm install pi-spi` first, and/or [override the npm dependency](http://stackoverflow.com/questions/15806152/how-do-i-override-nested-npm-dependency-versions)
- use node 0.10.x

Should be an easy fix here, but I'm in the middle of an overhaul and will need to switch back and double-check that the current branch of node-nrf works just fine with the pi-spi update first.

jlurgo, 2015-03-03:
Thank you Nathan!
I used node 0.10.36 and it worked!
We should keep this issue open till you fix it for the latest node version?
Thank you again!

natevw, 2015-03-03:
Glad to hear it! Yes, please leave the issue open until the currently published node-nrf gets an update/testing for this; thanks for bringing the trouble to my attention.

natevw, 2015-03-03:
Went ahead and only published nrf@0.8.3 from 60295b69403d7deed7b26aa110927e4c2ef394ba — to be honest, I cheated and only tested that 0.10.x still worked with that update *but* given my testing on pi-spi itself already, I'm pretty confident this library should now Just Work™ on a wider set of node engines too.

closed 2015-03-03 by natevw