natevw/node-nrf #37: Freezes on send

Lokki7, 2015-03-06:
Just started with test.js and stuck. I try to use Arduino as sender and RPi as receiver. 
RPi receive the first message and hangs. Here is debug log:
http://pastebin.com/10BanA4L

Tried C++ RF24 library - works
Tried reloading spi_bcm2708 - no effect

What I am doing wrong? Sorry for my english and thanks for advice!

natevw, 2015-03-09:
Did not get a chance to look at this over the weekend as I intended. Thanks for posting the debug log. I cannot see anything you are doing wrong.

The `sendPayload` code is stuck waiting for an interrupt to happen here, but it seems the IRQ pin does not fall again:
https://github.com/natevw/node-nrf/blob/98362357461865240bd4302aca2a9c5b39df3ee7/index.js#L340

I think the problem starts (here)[https://gist.github.com/Lokki7/0d566493fbd49dc09501#file-gistfile1-txt-L96] — but I am not sure what causes it yet. Maybe the response packet is getting sent too soon and the radio does not get a chance to reset IRQ?

Until I can try make this happen myself with hardware, you can try to wrap the `tx.write` response at [line 43 of test.js](https://github.com/natevw/node-nrf/blob/a36d6803321be60a8e38d9f3e84cf50678cddace/test.js#L43) in a little setTimeout to see if that works around the real problem?

Lokki7, 2015-03-10:
Right, there is no IRQ because no data actually sending. I happened to notice that the data is sent once if swap lines 182-183 in index.js. But the second attempt is still unsuccessful. So I think that there is something wrong with timeouts. Now I'm waiting for the delivery a few additional rf24 devices. When I get it I will try once again.

Lokki7, 2015-03-11:
Bad news - exactly the same behavior on another hardware. Wrapping in setTimeout also does not change anything.

natevw, 2015-03-11:
Sorry to hear that :-( Also interesting to hear that nothing actually sends to the Arduino. And even weirder that swapping the CE pulse logic changes things — just in case please double-check your CE pin connections and configuration. I know it works with RF24 though, so probably not it unless you tell it the wrong pin in your JS code.

May take you up on the offer if I can't reproduce this locally; I do finally have some hardware on order specifically for developing/testing this library — otherwise every time I find I've connected my Arduino and/or RasPi differently for some other project and that interrupts debugging.

(I hesitate to mention since I don't think it's worth your time yet, but FYI there's the "upcoming" branch at https://github.com/natevw/node-nrf/pull/32 — right now I'm still just fixing *new* bugs from moving code around, and if you can even get as far as this bug it will probably still be there for now.)

Lokki7, 2015-03-11:
Seems it was a software problem. After completely clean installation of Raspbian surprisingly it started to work. I will continue to install packages and if any of them will break nRF I will write about it. Thank you for your assistance!

Lokki7, 2015-03-11:
I rejoiced too early - it works only as "pong". "Ping" still not working.
Another strange behavior: no ack http://puu.sh/gw0YQ.png

natevw, 2015-03-12:
Interesting, thanks for reporting back! I'll close this for now then.

closed 2015-03-12 by natevw

shycherry, 2015-04-03:
Hello, I was running into the same freezes as @Lokki7 .
It was working before, I believe it stopped working after an apt-get (dist-??)upgrade ...
I've tried many node's versions, and even tried iojs 1.5... nothing changed: freeze after the first nrf reply.

I think the probleme is somehow related to that :
http://www.raspberrypi.org/forums/viewtopic.php?f=28&t=97314

So I managed to make it working again, without having to do a clean installation... here is what I've done :
sudo raspi-config
  Advanced Options -> Update ?
  Advanced Options -> SPI -> DISABLE
reboot
sudo raspi-config
  Advanced Options -> SPI -> ENABLE
reboot

@natevw thank you for this cool library :)

natevw, 2015-04-03:
Ah, yes! Those SPI changes might do it. Surprised there isn't a file I/O error when those devices aren't setup. Thanks @shycherry for sharing a simpler fix here!