natevw/node-nrf #38: event 'data' called twice?

lesion, 2015-04-10:
Hi,
I'm struggling with a little issue, hope you can help:

My setup is: an Arduino with  RF24 library and a simple ping test program with
this setup:
```
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xabcdabcd71 0xc2c2c2c2c2
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xabcdabcd71
RX_PW_P0-6	 = 0x20 0x00 0x00 0x00 0x00 0x00
EN_AA		 = 0x00
EN_RXADDR	 = 0x03
RF_CH		 = 0x48
RF_SETUP	 = 0x27
CONFIG		 = 0x0a
DYNPD/FEATURE	 = 0x03 0x06
Data Rate	 = 250KBPS
Model		 = nRF24L01+
CRC Length	 = 8 bits
PA Power	 = PA_MAX
```

and a Raspberry PI B+ that should receive, with node-nrf and this setup:
```
SPI device:	 /dev/spidev0.0
CE GPIO:	 22
IRQ GPIO:	 undefined
STATUS:		 0xe RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
Recommend use with IRQ pin, fallback handling is suboptimal.
RX_ADDR_P0–1:	 0xf0f0f0f0e1 0xabcdabcd71
RX_ADDR_P2–5:	 0xe2 0xe3 0xf1 0xf2
TX_ADDR:	 0xf0f0f0f0e1
RX_PW_P0–5:	 0x8 0x8 0x8 0x8 0x8 0x8
EN_AA:		 0x3f
EN_RXADDR:	 0x02
RF_CH:		 0x48
RF_SETUP:	 0x27
CONFIG:		 0x0b
DYNPD/FEATURE:	 0x02 0x07
Data Rate:	 250kbps
Model:		 nRF24L01+
CRC Length:	 8 bits
PA Power:	 PA_MAX
```

they are talking nice, but for each packet sent from arduino the 'data' event
on 'rx' pipe is called twice and the second time the `message` is the first char repeated N times
where N is the payload size.
so for example if the string `30921` is sent from arduino, on raspberry I see:
```
DATA EVENT 
30921   
DATA EVENT 
33333333
```
with this code:

```
  radio.channel(72)
    .dataRate('250kbps')
    .crcBytes(1)
    .transmitPower('PA_MAX')
    .autoRetransmit({count:15, delay:15000});

  radio.begin(function () {
    rx = radio.openPipe('rx', 0xABCDABCD71);
    rx.on('data',function(data){
      console.log("DATA EVENT ");
      console.log(data + ' ' + data.toString());
    });
  });
```

tried with RF24 C library and the packet is received correctly.
any suggestion?

sorry for my english and thanks for your work!

natevw, 2015-04-10:
This sounds a lot like https://github.com/natevw/node-nrf/issues/30, and I see you also are not using hardware IRQ pin? The workaround for now is to connect hardware IRQ pin, but I hope to fix the software fallback with the work in progress on https://github.com/natevw/node-nrf/pull/32

Thanks for the report, and please let me know if this can be closed as a duplicate of the other.

atomicus, 2015-07-29:
Yep, it's exactly what I came with in #30, just connect interrupt pin and will work fine.