natevw/node-nrf #39: Can't make a RPI chat with an Arduino

pirquessa, 2015-06-23:
Hi,

Thanks for your work on this lib !

I'm trying to make it works on a Raspberry PI 2 to chat with an Arduino for days now. I'm facing a problem with the RPI, I can't send data to the Arduino. I succeed to send data from the Arduino to the RPI.

I got the error "[Error: Packet timeout, transmit queue flushed.]" each time I try to write in the tx pipe.

You can find all my code here: https://github.com/pirquessa/RPI-and-Arduino-nRF24L01P.

Here are the details of the connection:

For the Arduino:
[image]

For the Raspberry:
[image]

Do you have any idea where can be the problem ?

Thanks a lot for your help.

natevw, 2015-06-23:
So it looks like the Raspberry Pi does receive from the Arduino, at least once? Be careful that the bytes are probably reversed (https://github.com/natevw/node-nrf/issues/3) but it looks like you are already reading as if Big Endian to flip back.

Your settings look correct, but it looks like neither the acks nor the writes from the RPi are getting back to the Arduino. Could you try a few things to debug:

- comment out https://github.com/pirquessa/RPI-and-Arduino-nRF24L01P/blob/55ea90843e3815a2b96d2abd4fc9ea8b722f0102/RaspberryPi%20code/nrf.js#L15 to not have the RPi respond right away in the TX. Does that make the Arduino *sending* ok?

- you can try uncomment https://github.com/pirquessa/RPI-and-Arduino-nRF24L01P/blob/55ea90843e3815a2b96d2abd4fc9ea8b722f0102/RaspberryPi%20code/nrf.js#L4 to get more details, although in this case most of the handling should be happening inside the nRF24 module unless it is getting switched into the wrong mode too early by the write, or something.

pirquessa, 2015-06-23:
Tanks a lot for this quick answer !

* In fact, when I comment the [line 15](https://github.com/pirquessa/RPI-and-Arduino-nRF24L01P/blob/55ea90843e3815a2b96d2abd4fc9ea8b722f0102/RaspberryPi%20code/nrf.js#L15) where I try to send data, I well receive all the messages. Once I try to send back the data, It stop with the error I mentioned. 

* I tried to use the debug thing but I don't understand any of the debug lines ... :( This is what I get when I'm only listening (line 15 commented):

```
Set CE low.
blocked for 130µs.
Set CE low.
blocked for 130µs.
execCommand FLUSH_TX 0
execCommand [ 'W_REGISTER', 5 ] [ 76 ]
execCommand FLUSH_RX 0
execCommand [ 'R_REGISTER', 6 ] 1
setStates { PWR_UP: true,
  PRIM_RX: false,
  EN_RXADDR: 0,
  RX_DR: true,
  TX_DS: true,
  MAX_RT: true,
  EN_DPL: true,
  EN_ACK_PAY: true,
  EN_DYN_ACK: true }
 - exec read: <SlowBuffer 42 05>
execCommand [ 'R_REGISTER', 6 ] 1
 - exec read: <SlowBuffer 4e 05>
execCommand [ 'R_REGISTER', 0 ] 1
 - exec read: <SlowBuffer 4e 0f>
execCommand [ 'R_REGISTER', 4 ] 1
 - exec read: <SlowBuffer 4e 03>
execCommand [ 'R_REGISTER', 0 ] 1
 - exec read: <SlowBuffer 4e 0f>
execCommand [ 'W_REGISTER', 0 ] [ 14 ]
blocked for 130µs.
execCommand [ 'W_REGISTER', 2 ] [ 0 ]
execCommand [ 'R_REGISTER', 7 ] 1
 - exec read: <SlowBuffer 4e 4e>
execCommand [ 'W_REGISTER', 7 ] [ 126 ]
execCommand [ 'R_REGISTER', 29 ] 1
 - exec read: <SlowBuffer 0e 07>
setStates { AW: 3,
  PRIM_RX: true,
  RX_ADDR_P1: <Buffer f0 f0 f0 f0 e1>,
  ERX_P1: true,
  ENAA_P1: true,
  DPL_P1: true }
setStates { AW: 3,
  RX_ADDR_P0: <Buffer f0 f0 f0 f0 d2>,
  ERX_P0: true,
  ENAA_P0: true,
  DPL_P0: true }
_checkStatus, irq = false checking = false
execCommand [ 'R_REGISTER', 0 ] 1
 - exec read: <SlowBuffer 0e 0e>
execCommand [ 'W_REGISTER', 0 ] [ 15 ]
blocked for 130µs.
execCommand [ 'R_REGISTER', 1 ] 1
 - exec read: <SlowBuffer 0e 3f>
execCommand [ 'R_REGISTER', 2 ] 1
 - exec read: <SlowBuffer 0e 00>
execCommand [ 'W_REGISTER', 2 ] [ 2 ]
execCommand [ 'R_REGISTER', 3 ] 1
 - exec read: <SlowBuffer 0e 03>
execCommand [ 'W_REGISTER', 11 ] <Buffer f0 f0 f0 f0 e1>
execCommand [ 'R_REGISTER', 28 ] 1
 - exec read: <SlowBuffer 0e 03>
Set CE true.
blocked for 130µs.
execCommand [ 'R_REGISTER', 1 ] 1
[... 374 lines cut ...]
execCommand [ 'R_REGISTER', 0 ] 1
 - exec read: <SlowBuffer 0e 0f>
execCommand [ 'W_REGISTER', 0 ] [ 14 ]
blocked for 130µs.
execCommand [ 'W_REGISTER', 10 ] <Buffer f0 f0 f0 f0 d2>
execCommand [ 'W_REGISTER', 16 ] <Buffer f0 f0 f0 f0 d2>
execCommand W_TX_PAYLOAD <Buffer 00 0c ef 5f>
Set CE true.
blocked for 10µs.
Set CE false.
blocked for 4µs.
_checkStatus, irq = true checking = false
execCommand [ 'R_REGISTER', 7 ] 1
 - exec read: <SlowBuffer 1e 1e>
gotStates { RX_P_NO: 7, TX_DS: 0, MAX_RT: 1, RX_DR: 0 } null
IRQ. { RX_P_NO: 7, TX_DS: 0, MAX_RT: 1, RX_DR: 0 }
execCommand FLUSH_TX 0
setStates { TX_DS: true, MAX_RT: true, RX_DR: false }
execCommand [ 'R_REGISTER', 7 ] 1
 - exec read: <SlowBuffer 1e 1e>
execCommand [ 'W_REGISTER', 7 ] [ 62 ]
Error sending:  [Error: Packet timeout, transmit queue flushed.]
```

I hope it will help you to understand my problem, I don't understand what it means :/

pirquessa, 2015-06-24:
Hi,

Since you supposed that the RF module could switch too early to write mode, I try to delay a little bit the tx.write like that:

```javascript
setTimeout(function(){
	tx.write(d);
}, 15);
```

It does not change anything :(

pirquessa, 2015-06-24:
I just add my wiring in [my repo](https://github.com/pirquessa/RPI-and-Arduino-nRF24L01P) if it can help...

pistoletov1974, 2015-06-24:
In my case (2 arduinos) and arduino+ cubieboard. Help capacitor on VCC and GND 47mF on nrf module

pirquessa, 2015-06-24:
Here my 2 rf modules are really close. I thought the capacitor would only help to improve range... Is that wrong?

pistoletov1974, 2015-06-27:
Hi! In my case with capasitor the modules become connected. Without capasitors was no ping on  approximately  1 meter distance.

pirquessa, 2015-07-01:
Well I tried my two modules on two Arduinos, one is still not able to send data. The problem must be with one of the module that can't send or the other that can't receive. I've ordered a new module and some capacitors, I'll try both solutions...

pirquessa, 2015-07-16:
Well, I finaly find some capacitor.

I went back to basic and test the modules with two arduinos (GettingStarted exemple). It really help to add capacitors (47uF) ! One of the module works well to send & receive data. The other send & receive too but when it send the data with "bool ok = radio.write( &time, sizeof(unsigned long) );", the bool ok is never true but the other module get the data and answer. It's strange.

pirquessa, 2015-07-16:
\o/ I can send data to the Arduino from the RPI ! \o/

Thanks again for your lib !

Is there a way to be sure that the data has been well received ? Sometime the Arduino fail to decode the data :(

closed 2015-07-16 by pirquessa