natevw/node-nrf #40: buffer => index out of range

gonAlonso, 2015-07-17:
Hi,
I've been trying to set an arduino and a RPi link without success. I've followed the Arduino example modified as @andreabellucci suggest on issue#25. Both my rf24 modules work using [tmrh20's RF24 lib](http://tmrh20.github.io/)
But if I use your test samples, the code crashes on pong mode or does not send at all on ping mode, at least my Arduino doesn't sense anything.
The error on pong mode is:
````
PONG back
buffer.js:494
    throw new RangeError('index out of range');
          ^
RangeError: index out of range
    at checkOffset (buffer.js:494:11)
    at Buffer.readUInt32BE (buffer.js:568:5)
    at PRX.<anonymous> (/srv/web/nodejs/rf24/git2/node-nrf/test.js:44:53)
    at PRX.emit (events.js:107:17)
    at readableAddChunk (_stream_readable.js:163:16)
    at PRX.Readable.push (_stream_readable.js:126:10)
    at /srv/web/nodejs/rf24/git2/node-nrf/xcvr_pipes.js:124:30
    at _finish (/srv/web/nodejs/node_modules/fifolock/index.js:31:13)
    at /srv/web/nodejs/node_modules/fifolock/index.js:36:20
    at /srv/web/nodejs/rf24/git2/node-nrf/xcvr_api.js:155:9
````
I've modified line 44 on test.js to look like:
````
console.log("Got data, will respond");    //, d.readUInt32BE(0));
````
Then it shows:
````
Got data, will respond
Got data, will respond
Error sending reply. [Error: Packet timeout, transmit queue flushed.]
````

I'm stuck! I'm not very good on Buffers on javascript. Any help?
Thank you

BTW, I've tested upcoming version without any luck.

natevw, 2015-07-17:
Thanks, yeah the upcoming branch is still in the middle of refactor (broken) and my youngest's evening routine many months has involved not very much laptop time for dad ;-)

Somehow maybe you are getting an empty buffer back, or one shorter than the 4 bytes needed to read a uint32? You could log:

    console.log("Got data, will respond", d.toString('hex'));

To see what the buffer contents are. It might help to see both the Arduino and JavaScript code you are using, if you can post it to https://gist.github.com/ — I wonder if perhaps the RF24 pinging sample you have is sending smaller packets than this library's pong sample expects?

gonAlonso, 2015-07-19:
Yeah! You hit the nail on the tail
The problem was the Arduino sending 1byte, and the javascript expecting 4bytes.
I've pasted my javascript code on https://gist.github.com/42fa206ba5e7bc95855e.git and my Arduino code on https://gist.github.com/e0d609c2530843906c58.git
Now, the next error. Arduino output:
````
RF24/examples/GettingStarted/

ROLE: Pong back
*** PRESS 'T' to begin transmitting to the other node

STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xf0f0f0f0d2 0xf0f0f0f0e1
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xf0f0f0f0d2
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x02
RF_CH		 = 0x4c
RF_SETUP	 = 0x27
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x3f 0x04
Data Rate	 = 250KBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
*** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK
Now sending "Hello - 1" as payload. failed.
...
````
And the RPi output
````
pi@raspberrypi /srv/web/nodejs/rf24/git/node-nrf $ sudo node test.js
SPI device:      /dev/spidev0.0
CE GPIO:         24
IRQ GPIO:        undefined
STATUS:          0xe RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
Recommend use with IRQ pin, fallback handling is suboptimal.
PONG back
RX_ADDR_P0–1:    0xf0f0f0f0d2 0xf0f0f0f0e1
RX_ADDR_P2–5:    0xc3 0xc4 0xc5 0xc6
TX_ADDR:         0xf0f0f0f0d2
RX_PW_P0–5:      0x20 0x20 0x0 0x0 0x0 0x0
EN_AA:           0x3f
EN_RXADDR:       0x03
RF_CH:           0x4c
RF_SETUP:        0x27
CONFIG:          0x0f
DYNPD/FEATURE:   0x03 0x07
Data Rate:       250kbps
Model:           nRF24L01+
CRC Length:      16 bits
PA Power:        PA_MAX
Got data, will respond 00000000000031202d206f6c6c6548
Got data, will respond 000000000000000000000000000000
Error sending reply. [Error: Packet timeout, transmit queue flushed.]
````
So, RPi receives the packet, but the Arduinio has no response
Note that Arduino has:
````
 radio.setAutoAck(1);                    // Ensure autoACK is enabled
````
So I assume the rf24 module expects an ACK from RPi on every sent packet. How do I force autoAck on the RPi side? Is it the problem, anyway?
Moreover, it seems that RPi is sending twice the response.

gonAlonso, 2015-07-26:
OK, I've managed to send and receive packets from both sides. Now I've encountered the reversed order issue, solved with __split("").reverse().join("")__ on the RPi side, at least with string packets, must see what happens with floats and so on.
The problem I'm facing now is an autoAck misconfiguration. I'll close this issue now.
Thank you Nathan!

closed 2015-07-26 by gonAlonso