natevw/node-nrf #41: Unable to configure Pi

napikwaan, 2015-08-09:
Hey Nate.  I've been trying to get up and running with this, but after running `radio.printDetails()` it seems like I'm not able to configure things correctly.  Running node v0.12.0, CSE pin = 22, IRQ pin = 7.  spidev0.0 and spidev0.1 have been enabled.  The code I'm running is a modification of the example in the README:
```javascript
var spiDev = '/dev/spidev0.0',
    cePin  = 22,
    irqPin = 7,
//    pipes  = [0xF0F0F0F0E1, 0xF0F0F0F0D2];
    pipes  = [Buffer('F0F0F0F0E1', 'hex'), Buffer('F0F0F0F0D2', 'hex')];

var radio = require('nrf').connect(spiDev, cePin, irqPin);
radio.channel(0x4c).dataRate('1Mbps').crcBytes(2).autoRetransmit({ count: 15, delay: 4000 });

radio.begin(function() {
  var rx = radio.openPipe('rx', pipes[0]),
      tx = radio.openPipe('tx', pipes[0]);

  radio.printDetails();

  rx.on('data', function(d) {
    console.log(d);
  });

  tx.write('Testing');
});
```

`radio.printDetails()` gives me a ton of 0's:

Key | Value
------------ | -------------
SPI device: | /dev/spidev0.0
CE GPIO: | 22
IRQ GPIO: | 7
STATUS:	 | 0x0 RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=0 TX_FULL=0
RX_ADDR_P0–1: | 0x0000000000 0x0000000000
RX_ADDR_P2–5: | 0x00 0x00 0x00 0x00
TX_ADDR: | 0x0000000000
RX_PW_P0–5: | 0x0 0x0 0x0 0x0 0x0 0x0
EN_AA: | 0x00
EN_RXADDR: | 0x00
RF_CH: | 0x0
RF_SETUP: | 0x00
CONFIG: | 0x00
DYNPD/FEATURE:	 | 0x00 0x00
Data Rate: | 1Mbps
Model: | nRF24L01
CRC Length: | Disabled
PA Power: | PA_MIN

Additionally, the following message is logged to the console:
> Unexpected IRQ during transmit phase!

natevw, 2015-08-11:
First thing that comes to mind is, have you made sure the pins you are using the correct ones? The [logical pin numbering](http://pi.gadgetoid.com/pinout) of the Raspberry Pi is different than the physical order of pins. For example, logical GPIO 22 is what would be traditionally called pin 15 of the header itself.

napikwaan, 2015-08-13:
Thanks for the reply.  When referencing pin numbers in my previous comment I was referring to the GPIO pins.  My pins are wired like this:

nRF24L01 | Pi
------------- | -------------
VCC | 3.3v (17)
GND | GND (9)
CSN | GPIO 8 (24)
CE | GPIO 22 (15)
MOSI | GPIO 10 (19)
SCK | GPIO 11 (23)
MISO | GPIO 9 (21)
IRQ | GPIO 7 (26)

Incidentally, I have been been able to successfully test the nRF24L01 using the instructions on [HomeAutomationForGeeks](http://www.homeautomationforgeeks.com/rf24hardware.shtml) so I know the unit is functional.  I'm sure there's something basic I'm missing here, I just haven't been able to figure it out yet.

natevw, 2015-08-13:
Ah! You might be running into an issue where the C++ RF24 library disables the kernel SPI driver's select line. Could you follow the instructions at https://github.com/natevw/node-nrf#node-nrf-or-pi-spi-not-working-after-using-c-rf24-library and see if that helps?

napikwaan, 2015-08-16:
That did the trick.  When running `sudo modprobe spi_bcm2708` I got an error, but rebooting the Pi got it working.  Thanks a ton for your help, natevw.

closed 2015-08-16 by napikwaan