natevw/node-nrf #44: SPI random crash "Unexpected IRQ during transmit phase!"

belsource, 2015-12-16:
I tried:
>sudo modprobe -r spi_bcm2708
>sudo modprobe spi_bcm2708

 but actually I have snd_bcm2835 module (Raspbian on RPi2), so I used snd_bcm2835 in command

Here is my problem:
![Image of issue]
(https://habrastorage.org/files/4a5/375/6b5/4a53756b5b16456ca4c54a55d2c1e487.png)

belsource, 2015-12-22:
Script works 1-10 hours and then:

>fs.js:500
 > return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
  >               ^
>Error: EACCES, permission denied '/sys/class/gpio/gpio25/value'
 >   at Error (native)
 >  at Object.fs.openSync (fs.js:500:18)
 >  at Object.exports.connect (/var/www/html/node_modules/pi-pins/index.js:29:21)
 >  at Object.exports.connect (/var/www/html/nodejs/index.js:32:19)
 > at Object.<anonymous> (/var/www/html/nodejs/server.js:8:11)
 >  at Module._compile (module.js:460:26)
 >  at Object.Module._extensions..js (module.js:478:10)
 >  at Module.load (module.js:355:32)
 >  at Function.Module._load (module.js:310:12)
 >  at Function.Module.runMain (module.js:501:10)
>error: Forever detected script exited with code: 1
>error: Script restart attempt #1
>fs.js:500
> return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
  >               ^
>Error: EACCES, permission denied '/sys/class/gpio/gpio7/value'
  >  at Error (native)
  > at Object.fs.openSync (fs.js:500:18)
 >  at Object.exports.connect (/var/www/html/node_modules/pi-pins/index.js:29:21)
 >   at Object.exports.connect (/var/www/html/nodejs/index.js:33:46)
 >   at Object.<anonymous> (/var/www/html/nodejs/server.js:8:11)
 >   at Module._compile (module.js:460:26)
 >   at Object.Module._extensions..js (module.js:478:10)
 >   at Module.load (module.js:355:32)
 >   at Function.Module._load (module.js:310:12)
 >   at Function.Module.runMain (module.js:501:10)
>error: Forever detected script exited with code: 1

natevw, 2015-12-24:
Hmm, interesting. The error messages really point to a problem with GPIO access (for which this library uses my separate [pi-pins](https://github.com/natevw/pi-pins) module) rather than SPI, so I guess it's not too surprising that the kernel module trick does not make a difference.

However, it looks you are trying to use GPIO07 for your IRQ pin? This is the SPI_CE0_N pin for /dev/spidev0.0 so I suspect this could be part of the problem. Which pins is your nRF24 module connected to on the RasPi header? You should have a separate pin for IRQ than for its CSN (which should be SPI_CE0_N which is physical pin 24 on all models afaik:

[image]

Hope this helps!

belsource, 2015-12-25:
So, should I remove nrf24 IRQ pin?

[image]

belsource, 2015-12-25:
Without IRQ <-> GPIO7 - it doesn't work at all

belsource, 2015-12-29:
I still dont unserstand what is happening.. nodejs crashes every 1-24 hours randomly

natevw , what do u use  for nodejs background process? I use "forever" ...

natevw, 2016-01-01:
I do not recommend disconnecting IRQ pin, it currently can lead to strange issues e.g. #30. I may have spoken too soon, flipping between pinout diagrams and got confused — your original pinouts may not necessarily have been wrong. Somebody else just filed https://github.com/natevw/pi-spi/issues/19 so I wonder if there is some issue with the new kernel drivers? [Note that that is a SPI issue and this is a GPIO issue though.] Or are you sure you don't have other processes that are occasionally trying to access GPIO and conflicting somehow?

You can use whatever you want for process restarts, forever or node-supervisor or whatever and in theory it shouldn't make a difference to this issue. (I tend to use node-foreman and node-supervisor in development, Ubuntu's [upstart](http://upstart.ubuntu.com/) in production and something like `while sleep 1; node project/myscript; done;` for personal stuff when I'm too hurried to find back the upstart documentation.)