natevw/node-nrf #45: Connection between Arduino and Raspberry PI seams impossible (but it may be my own newbieness)

knoid, 2015-12-22:
Hi natevw, thanks for the great library!

I'm trying to connect an arduino and a raspberry PI wirelessly but I obviously can't do it. I have really simple programs in both devices. I've tried with and without capacitors (100uf, 220uf), different configurations, I can't think of anything else. I have an arduino pro mini (if that makes a difference). I guess they are properly connected because the output configuration looks right.

I've checked the spi module loading on the rpi and it loads `spi_bcm2835`. Would that work?

A sample program to test with would work for me because I couldn't find an example for raspberry in nodejs and arduino in C; but meanwhile, this is what I got.

Code in raspberry pi:

```
var NRF24 = require('nrf'),

spiDev = '/dev/spidev0.0',
cePin = 24, irqPin = 25,

pipes = [0xF0F0F0F0E1, 0xF0F0F0F0D2],

nrf = NRF24.connect(spiDev, cePin, irqPin);
nrf.
  channel(0x4c).
  transmitPower('PA_LOW').
  dataRate('1Mbps').
  crcBytes(2).
  begin(() => {
    var tx = nrf.openPipe('tx', pipes[1]),
        rx = nrf.openPipe('rx', pipes[0]);
    rx.on('data', (d) => {
      console.log('Radio data:', d.readUInt32BE(0));
    });
    tx.on('error', (err) => {
      console.log(err);
    });
    tx.on('ready', () => {
      nrf.printDetails(() => {
        setInterval(() => {
          console.log('sending data');
          var b = new Buffer(8);
          b.writeUInt32BE(42, 0);
          tx.write(b);
        }, 1000);
      });
    });
  });
```

Output:
```
SPI device:	 /dev/spidev0.0
CE GPIO:	 24
IRQ GPIO:	 25
STATUS:		 0xe RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0–1:	 0xf0f0f0f0d2 0xf0f0f0f0e1
RX_ADDR_P2–5:	 0xc3 0xc4 0xc5 0xc6
TX_ADDR:	 0xf0f0f0f0d2
RX_PW_P0–5:	 0x0 0x0 0x0 0x0 0x0 0x0
EN_AA:		 0x3f
EN_RXADDR:	 0x03
RF_CH:		 0x4c
RF_SETUP:	 0x03
CONFIG:		 0x0f
DYNPD/FEATURE:	 0x03 0x07
Data Rate:	 1Mbps
Model:		 nRF24L01+
CRC Length:	 16 bits
PA Power:	 PA_LOW
sending data
[Error: Packet timeout, transmit queue flushed.]
sending data
[Error: Packet timeout, transmit queue flushed.]
sending data
[Error: Packet timeout, transmit queue flushed.]
^C
```

Code in arduino:
```
#include <SPI.h>
#include "RF24.h"
#include "printf.h"

RF24 radio(10, 5);

const uint64_t pipes[2] = { 0xF0F0F0F0E1LL, 0xF0F0F0F0D2LL };

void setup() {
  printf_begin();
  Serial.begin(19200);

  radio.begin();
  radio.setChannel(0x4c);
  radio.setCRCLength(RF24_CRC_16);
  radio.setPALevel(RF24_PA_LOW);

  radio.openWritingPipe(pipes[0]);
  radio.openReadingPipe(1, pipes[1]);

  radio.printDetails();

  radio.startListening();
  Serial.println("startListening");
}

void loop() {
  unsigned long got_time = 0;

  if (radio.available()) {
    // Variable for the received timestamp
    while (radio.available()) {
      radio.read( &got_time, sizeof(unsigned long) );
    }

    radio.stopListening();
    radio.write( &got_time, sizeof(unsigned long) );
    radio.startListening();
    Serial.print(F("Sent response "));
    Serial.println(got_time);
  }
}
```

Output:
```
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xf0f0f0f0e1 0xf0f0f0f0d2
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xf0f0f0f0e1
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x03
RF_CH		 = 0x4c
RF_SETUP	 = 0x03
CONFIG		 = 0x0e
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_LOW
startListening
```

belsource, 2015-12-22:
You are very close to the goal.
If u are using RPi2, then try to use cePin = 25, irqPin = 7 - it works for me (IRQ - the last 8 pin, in manuals is not used)

knoid, 2015-12-23:
I'm using a Raspberry Pi Model B Revision 2.0 (512MB) and I understand that pin 25 is GND, so that wouldn't work.

jstnjns, 2015-12-23:
@knoid I believe @belsource meant GPIO 25 (pin 22)?  [image]

natevw, 2015-12-24:
Yeah, I first thought maybe your RX/TX addresses were swapped but looking again I think you have it correct. @jstnjns is probably on to something, a common trip-up that I need to document better:

node-nrf (well, [pi-pins](https://github.com/natevw/pi-pins) really) uses the "virtual" GPIO numbers, not the physical pins on the header. So please make sure that if you have `cePin = 24, irqPin = 25` in your code that you plug in to pins 18 (GPIO24) and 22 (GPIO25) on the header.

knoid, 2015-12-24:
Thanks for the tips, I've checked the GPIOs and it all seems to be right. I'm closing this issue for now because the problem may be the arduino's power so I will continue trying when I get my hands on a proper voltage regulator.

closed 2015-12-24 by knoid

natevw, 2015-12-24:
No worries, hope you get it sorted or otherwise feel free to ping back here. The nRF24 needs to TX to ack the packets, which could be a significant current draw change if you're e.g. just using a resistor to get to 3.3v or something.