natevw/node-nrf #48: "Unexpected IRQ during transmit phase!" with ack payload

wuyuanyi135, 2016-02-24:
Hi,
[Code](https://github.com/natevw/node-nrf/blob/master/index.js#L344)
My code uses ack payload. Therefore, I do not expect `TX_DS` after `tx.write`. At PRX, when I call `tx.write`, the callback actually gets the flag of `RX_DR`, which resulting in the false warning.

The fact is that, after calling `tx.write`, the event listener is expected to be triggered right after the packet is "sent". Despite the other two commands (`W_TX_PAYLOAD`, `W_TX_PAYLOAD_NOACK`), this `W_ACK_PAYLOAD` does not actually fire an interruption. Instead, the following signal is `RX_DR`, when the PTX sends a packet. Then the `TX_DR` or `MAX_RT` should be set due to the acknowledge payload sent. See the figure in pg.45.

[image]

In addition, I noticed the [line 348](https://github.com/natevw/node-nrf/blob/master/index.js#L348), where you leave the `RX_DR` flag untouched. Is there any specific reason to do that?

Thank you

wuyuanyi135, 2016-02-24:
I have fixed this in my repo. Will submit a PR

pirquessa, 2016-10-23:
I have the problem of "Unexpected IRQ during transmit phase!".

I tried your version of the lib but I don't have any exchanges between my devices... At least with the original version I have some exchanges before a crash.

Did we need to change anything else than the main index.js file ?