natevw/node-nrf #51: Is it possible to catch error when MAX_RT asserting?

wuyuanyi135, 2016-03-03:
I found that the the MAXRT event is registered in the 'interrupt' event callback. I have no idea how can I catch the error. I feel that using pipe.emit() would be better way.

natevw, 2016-03-03:
You should be able to use the usual node.js `.on('error', function (e) { /* your handler */ })` on the pipe which you are writing to. Handling the error will prevent your process from exiting, although note that any additional queued packets will be dropped, so you may need to do some recovery handling of your own depending on your protocol.

Feel free to re-open if I misunderstood the question, or that isn't working…otherwise I'm assuming this answers it.

closed 2016-03-03 by natevw

wuyuanyi135, 2016-03-04:
But my process still quits
```js
PxX.prototype.willWriteAck = function (buffer) {
        var self = this;
        return new Promise (function (resolve, reject) {
            self.write(buffer);
            self.once("error", function(d) {
                console.log("caught");
            });
            self.once("data", function(d) {
                resolve (d);
            });
        });
    }
```

This code still quit with 

```
Error: Packet timeout, transmit queue flushed.
    at /home/pi/wireless/node_modules/nrf/index.js:360:24
    at /home/pi/wireless/node_modules/nrf/index.js:81:25
```
Note: the line number may not be consistent

wuyuanyi135, 2016-03-04:
Dah, sorry, I invoked another function. `.on('error', function(){})` actually work perfectly.