natevw/node-nrf #52: node-nrf silent

frinux, 2016-03-03:
Hi,
I'm trying to set node-nrf on my RaspPi B+, and I can't get it work. The hardware is OK, I used the C++ example code to check:
```
pi@raspberrypi ~/nrf24l01 $ sudo ./rx
================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO22
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x3e
EN_RXADDR	 = 0x3f
RF_CH		 = 0x5a
RF_SETUP	 = 0x07
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x3f 0x04
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
Received payload at 2130902 : 18.400000°C, 62.799999 percent
aspberrypi ~/nrf24l01 $ sudo ./rx
================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO22
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x3e
EN_RXADDR	 = 0x3f
RF_CH		 = 0x5a
RF_SETUP	 = 0x07
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x3f 0x04
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
Received payload at 2130902 : 18.400000°C, 62.799999 percent
```

And then I clean with 
```
pi@raspberrypi ~/nrf24l01_nodejs $ sudo modprobe -r spi_bcm2835
pi@raspberrypi ~/nrf24l01_nodejs $ sudo modprobe spi_bcm2835
(or reboot)
```

My node code is very simple:

```
var spiDev = "/dev/spidev0.0";
var cePin = 15; //RPI_BPLUS_GPIO_J8_15
var irqPin = null;
var channel = 0x5a; //90

var radio = require('nrf').connect(spiDev, cePin, irqPin);

radio
	.channel(channel)
	.dataRate('1Mbps')
//	.crcBytes(2)
//	.autoRetransmit({count:15, delay:4000})
	;

radio.begin(function () {
    var rx = radio.openPipe('rx', 0xCCCCCCCCC3);
	rx.pipe(process.stdout);
});
spiDev = "/dev/spidev0.0";
var cePin = 15; //RPI_BPLUS_GPIO_J8_15
var irqPin = null;
var channel = 0x5a; //90

var radio = require('nrf').connect(spiDev, cePin, irqPin);

radio
	.channel(channel)
	.dataRate('1Mbps')
//	.crcBytes(2)
//	.autoRetransmit({count:15, delay:4000})
	;

radio.begin(function () {
    var rx = radio.openPipe('rx', 0xCCCCCCCCC3);
	rx.pipe(process.stdout);
});
```

But the console is always silent...

What am I doing wrong?

Thanks

maximov-ru, 2016-03-12:
you need something as:
```
var tx = radio.openPipe('tx', 0xCCCCCCCCC3,{size: 'auto',autoAck:false});
tx.on('ready', function () {
  tx.write(process.stdout);
});
```

frinux, 2016-03-13:
@maximov-ru why tx? It is normally used to transmit data, and  all I want is receive data.
However, I tried but console is always silent: 

```
var tx = radio.openPipe('tx', 0xCCCCCCCCC3);
tx.pipe(process.stdou
```t);