natevw/node-nrf #57: Problems getting node-nrf on Rasp Pi to talk to tmrh20 on Arduino Uno.

runia1, 2016-11-15:
My goal is being able to send commands from my nodejs script and have the arduino pick up the command and do something.  As a proof of concept, I'm just trying to get the Arduino to print out a message when data comes in.

**My devices:**

1. Raspberry pi 1B (Running this library on Node v6.9.1)

```
var NRF24 = require('nrf');
var spiDev = "/dev/spidev0.0";
//pins for rasp pi 1B according to: https://gist.github.com/natevw/5789019
var cePin = 18, irqPin = 22;

var radio = NRF24.connect(spiDev, cePin, irqPin);
radio.channel(0x4c)
  .transmitPower('PA_MAX')
  .dataRate('1Mbps')
  .crcBytes(2)
  .autoRetransmit({count:5, delay:4000})
  .begin(function () {
    //open pipe for sending
    var tx = radio.openPipe('tx', 0xD2F0F0F0F0);

    //when the transmitter is ready...
    tx.on('ready', function () {
      //radio.printDetails();
      for(var i=1; i<=100; i++) {
        var data = 'sending ' + i;

        console.log(data);
        tx.write(data);
      }
    });
  });
```
This runs without errors, and appears to send out data but I never receive anything on my Arduino.  When I uncomment the printDetails() I get this:
```
SPI device:      /dev/spidev0.0
CE GPIO:         18
IRQ GPIO:        22
STATUS:          0xe RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0–1:    0xd2f0f0f0f0 0x65646f4e31
RX_ADDR_P2–5:    0xc3 0xc4 0xc5 0xc6
TX_ADDR:         0xd2f0f0f0f0
RX_PW_P0–5:      0x0 0x0 0x0 0x0 0x0 0x0
EN_AA:           0x3f
EN_RXADDR:       0x01
RF_CH:           0x4c
RF_SETUP:        0x07
CONFIG:          0x0e
DYNPD/FEATURE:   0x03 0x07
Data Rate:       1Mbps
Model:           nRF24L01+
CRC Length:      16 bits
PA Power:        PA_MAX
```
Which leads me to believe the Pi has a good connection to the NRF chip (It's able to assign the address and other configs.)

2. Arduino Uno R3 (Running tmrh20 c++ lib)
```
#include <printf.h>
#include <SPI.h>
#include "RF24.h"

/* Hardware configuration: Set up nRF24L01 radio on SPI bus plus pins 7 & 8 */
RF24 radio(7,8);

uint64_t recieveAddress = 0xD2F0F0F0F0LL;

void setup() {
  Serial.begin(115200);
  printf_begin();
  Serial.println(F("Arduino is being setup()."));

  radio.begin();
  radio.setChannel(0x4c);           //Physical channel: 0x4c, AKA: 76
  radio.setPALevel(RF24_PA_MAX);    //Transmit Power: MAX
  radio.setDataRate(RF24_1MBPS);    //Data Rate: 1Mbps
  radio.setCRCLength(RF24_CRC_16);  //CRC Length: 16 bit, AKA: 2 bytes.
  radio.setRetries(15, 5);         //Auto-retransmit: 5 times, once every 4000us.
  radio.setAutoAck(true);

  radio.openReadingPipe(1, recieveAddress);
  
  // Start the radio listening for data
  radio.startListening();

  Serial.println(F("Arduino finished setup. Radio is listening."));
  radio.printDetails();
}

void loop() {
  unsigned long data;
  
  if(radio.available()) {
    //radio.read(&data, sizeof(data));

    //JUST PRINT OUT THAT THERE IS DATA TO READ, WE DON'T CARE WHAT IT IS YET.
    Serial.print(F("Recieving data: "));
    //Serial.println(data);
  }

}
```

Results of the printDetails()
```
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0x65646f4e31 0xd2f0f0f0f0
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0x65646f4e31
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x02
RF_CH		 = 0x4c
RF_SETUP	 = 0x03
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01+
  CRC Length	 = 16 bits
PA Power	 = PA_LOW
```
So I'm pretty sure the wiring is correct on the arduino as well, as I'm able to set the address and other configs.

**Troubleshooting I've tried**
- Arduino Uno to another Uno both with same wiring and both running the tmrh20 Lib.  These two were able to talk back and forth, so I'm pretty certain I have the wiring right and that I'm using the library right on the Uno's end.

Any help is appreciated, I'm not really sure where to go from here.  Maybe I'm not writing to the tx stream correctly?  I'm new to nodejs streams but I'm pretty sure you just have to do a `tx.write("some data");` to send out some data correct?

runia1, 2016-11-15:
Also I haven't installed the c++ lib on the Raspberry Pi so that known bug shouldn't be an issue.

HapCoderWei, 2017-06-23:
I have same problem as yours, and don't know what is wrong. Below is my printDetails display:
```
RX Ready
SPI device:	 /dev/spidev0.0
CE GPIO:	 25
IRQ GPIO:	 24
STATUS:		 0xe RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0–1:	 0xe7e7e7e7e7 0x3443101001
RX_ADDR_P2–5:	 0xc3 0xc4 0xc5 0xc6
TX_ADDR:	 0xe7e7e7e7e7
RX_PW_P0–5:	 0x0 0x0 0x0 0x0 0x0 0x0
EN_AA:		 0x3f
EN_RXADDR:	 0x03
RF_CH:		 0x4c
RF_SETUP:	 0x0d
CONFIG:		 0x0f
DYNPD/FEATURE:	 0x03 0x07
Data Rate:	 2Mbps
Model:		 nRF24L01+
CRC Length:	 16 bits
PA Power:	 PA_HIGH
```
After checked the resisters in list, I think maybe the `RX_PW_P0–5:      0x0 0x0 0x0 0x0 0x0 0x0` is the reason. In general, the RX_PW_P0 is payloadWidth, and it should be a value range 1~32, but here it is 0, which is illegal. But I don't know how to set it in nodejs. Anyone can help?

natevw, 2017-06-23:
@HapCoderWei Thanks for the report. I think if you are using the dynamic payloads you do not need to worry about RX_PW_Pn registers but you can set a fixed size when creating your pipe e.g. `radio.openPipe('rx', "0xe7e7e7e7e7", {size:16})` to see if that helps?

ipatalas, 2017-12-22:
@runia1 @HapCoderWei anyone of you got this solved? Just spent a day troubleshooting the same...
Everything seems to be correct, but it doesn't work. I've got opposite situation (listening on RPi and transmitting from Arduino) but tried yours and it still doesn't work.

RPi registers (listener):
```
SPI device:      /dev/spidev0.0
CE GPIO:         22
IRQ GPIO:        18
listening...
STATUS:          0xe RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0–1:    0xe7e7e7e7e7 0x7465737431
RX_ADDR_P2–5:    0xc3 0xc4 0xc5 0xc6
TX_ADDR:         0xe7e7e7e7e7
RX_PW_P0–5:      0x0 0x0 0x0 0x0 0x0 0x0
EN_AA:           0x3f
EN_RXADDR:       0x02
RF_CH:           0x4c
RF_SETUP:        0x27
CONFIG:          0x0b
DYNPD/FEATURE:   0x02 0x07
Data Rate:       250kbps
Model:           nRF24L01+
CRC Length:      8 bits
PA Power:        PA_MAX
```

Arduino (transmitter):
```
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0x7465737431 0xd2f0f0f0f0
RX_ADDR_P2-5     = 0xc3 0xc4 0xc5 0xc6
TX_ADDR          = 0x7465737431
RX_PW_P0-6       = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA            = 0x3f
EN_RXADDR        = 0x02
RF_CH            = 0x4c
RF_SETUP         = 0x27
CONFIG           = 0x0a
DYNPD/FEATURE    = 0x3f 0x04
Data Rate        = 250KBPS
Model            = nRF24L01+
CRC Length       = 8 bits
PA Power         = PA_MAX
```

natevw, 2017-12-23:
I don't see anything obviously wrong based on your config (see notes below). You might try set `radio._debug = true` right away and see if that gives any clues?

---

On your RasPi you have:

```
CONFIG:          0x0b
DYNPD/FEATURE:   0x02 0x07
```

Config is `EN_CRC | PWR_UP | PRIM_RX`, dynamic payloads are `DPL_P1`, and feature is `EN_DPL | EN_ACK_PAY | EN_DYN_ACK`

On the Arduino you have:

```
CONFIG           = 0x0a
DYNPD/FEATURE    = 0x3f 0x04
```

Config is `EN_CRC | PWR_UP`, dynamic payloads are enabled on all 5 pipes, and feature is `EN_DPL`.

ipatalas, 2017-12-23:
Thanks for explaining what lies beneath the config register and those other two. 
I did notice they differ but couldn't understand what these values mean.
Gonna give that debug flag a shot but it's Christmas time now :santa: :christmas_tree:
I will post an update around Tuesday when I will be able to test it again.

ipatalas, 2017-12-26:
Turning debug mode doesn't ring any bells for me. 
```
setStates { RF_CH: 76 }
setStates { RF_DR_LOW: true, RF_DR_HIGH: false }
setStates { EN_CRC: true, CRCO: 0 }
Set CE low.
blocked for 130µs.
Set CE low.
blocked for 130µs.
execCommand FLUSH_TX 0
execCommand [ 'W_REGISTER', 5 ] [ 76 ]
execCommand FLUSH_RX 0
execCommand [ 'R_REGISTER', 6 ] 1
setStates { PWR_UP: true,
  PRIM_RX: false,
  EN_RXADDR: 0,
  RX_DR: true,
  TX_DS: true,
  MAX_RT: true,
  EN_DPL: true,
  EN_ACK_PAY: true,
  EN_DYN_ACK: true }
 - exec read: <Buffer 0e 27>
execCommand [ 'R_REGISTER', 0 ] 1
 - exec read: <Buffer 0e 0b>
execCommand [ 'R_REGISTER', 0 ] 1
 - exec read: <Buffer 0e 0b>
execCommand [ 'W_REGISTER', 0 ] [ 10 ]
blocked for 130µs.
execCommand [ 'W_REGISTER', 2 ] [ 0 ]
execCommand [ 'R_REGISTER', 7 ] 1
 - exec read: <Buffer 0e 0e>
execCommand [ 'W_REGISTER', 7 ] [ 126 ]
execCommand [ 'R_REGISTER', 29 ] 1
 - exec read: <Buffer 0e 07>
setStates { AW: 3,
  PRIM_RX: true,
  RX_ADDR_P1: <Buffer 74 65 73 74 31>,
  ERX_P1: true,
  ENAA_P1: true,
  DPL_P1: true }
execCommand [ 'R_REGISTER', 0 ] 1
_checkStatus, irq = false checking = false
_checkStatus, irq = true checking = true
 - exec read: <Buffer 0e 0a>
execCommand [ 'W_REGISTER', 0 ] [ 11 ]
blocked for 130µs.
execCommand [ 'R_REGISTER', 1 ] 1
 - exec read: <Buffer 0e 3f>
execCommand [ 'R_REGISTER', 2 ] 1
 - exec read: <Buffer 0e 00>
execCommand [ 'W_REGISTER', 2 ] [ 2 ]
execCommand [ 'R_REGISTER', 3 ] 1
 - exec read: <Buffer 0e 03>
execCommand [ 'W_REGISTER', 11 ] <Buffer 74 65 73 74 31>
execCommand [ 'R_REGISTER', 28 ] 1
 - exec read: <Buffer 0e 02>
Set CE true.
blocked for 130µs.
execCommand [ 'R_REGISTER', 7 ] 1
 - exec read: <Buffer 0e 0e>
gotStates { RX_P_NO: 7, TX_DS: 0, MAX_RT: 0, RX_DR: 0 } null
execCommand [ 'R_REGISTER', 7 ] 1
 - exec read: <Buffer 0e 0e>
gotStates { RX_P_NO: 7, TX_DS: 0, MAX_RT: 0, RX_DR: 0 } null
IRQ. { RX_P_NO: 7, TX_DS: 0, MAX_RT: 0, RX_DR: 0 }
```

Based on the link you provided it seems like it's in RX mode which is good (_PRIM_RX in CONFIG register is 1 (receiver enabled), CE is high.  The radio can receive packets._).

I've been trying today for another couple of hours and failed again :( Tried different settings (data rate, crc, PA level). Checked the other chip select pin on RPi or even completely replaced RF24 module with another one (have some spare items) but still no luck.
I really wish I knew what am I doing wrong. 
I'm gonna borrow another NodeMCU from a friend next week to see how it goes for two same chips.

One side of my connection just in case somebody can spot something I wasn't able to: 
[image]
(NodeMCU pins for reference: https://iotbytes.wordpress.com/nodemcu-pinout/)

natevw, 2017-12-29:
Not seeing anything obvious, but it's been a bit since I had one set up myself. Hope to take a closer look soon.

ipatalas, 2018-01-08:
I've been finally able to set up two NodeMCUs with nrf modules and they work just fine. Connection is not very reliable, but I'm getting quite some responses for an example app.
That makes me believe this side of the connection is done properly and radio modules themselves are functional.

Here is the other side:
[image]
Still not even close to figure out what might be wrong here.