natevw/node-nrf #60: Install error

djdjocker, 2017-12-18:
Hello,

I'm currently trying to install this node module and get the following errror on my RPI 2 model B with Jessy installed :

`npm WARN deprecated queue-async@1.0.7: renamed to d3-queue

> pi-spi@1.0.2 install /home/pi/node_modules/nrf/node_modules/pi-spi
> node-gyp rebuild

gyp: Call to 'node -e "require('nan')"' returned exit status 1. while trying to load binding.gyp
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/share/node-gyp/lib/configure.js:344:16)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.9.35+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/node_modules/nrf/node_modules/pi-spi
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian

> epoll@0.1.22 install /home/pi/node_modules/nrf/node_modules/pi-pins/node_modules/epoll
> node-gyp rebuild

node.js:815
    var cwd = process.cwd();
                      ^
Error: ENOENT, no such file or directory
    at Function.startup.resolveArgv0 (node.js:815:23)
    at startup (node.js:58:13)
    at node.js:906:3
npm ERR! pi-spi@1.0.2 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the pi-spi@1.0.2 install script.
npm ERR! This is most likely a problem with the pi-spi package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls pi-spi
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 4.9.35+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "nrf"
npm ERR! cwd /home/pi
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian

npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/npm-debug.log
npm ERR! not ok code 0`

Is Anyone can help me pleaze ?

natevw, 2017-12-19:
> npm WARN This failure might be due to the use of legacy binary "node"
> npm WARN For further explanations, please read /usr/share/doc/nodejs/README.Debian

This does look like a pretty old version of node, can you try 0.12 at least? Or you might also look into https://stackoverflow.com/questions/21168141/cannot-install-packages-using-node-package-manager-in-ubuntu if that seems relevant — I think some distros don't like to install node under its normal name and that might be confusing node-gyp.

djdjocker, 2017-12-28:
Hello, Thank you.

My main problem was due to bad installation of Node Js.

closed 2017-12-28 by djdjocker

natevw, 2017-12-29:
Glad you got it working!