natevw/node-nrf #61: Nrf24 module doesn't communicate

luckysistemi, 2018-02-01:
Hi, I'm trying to work with your library without success.
I Have a sensor based on Arduino nano that communicates with a Raspberry.
Arduino and Raspberry are working well with a C++ program, that I need to move to javascript.
I don't know what I'm doing wrong or what I'm not doing, but the two are absolutely not communicating.
As You mention in the documentation, I need to reload the SPI driver with 
sudo modprobe -r spi_bcm2835
sudo modprobe spi_bcm2835
and this enables the communication between Raspberry and RF24 module.

The fact that I have a working project tells me that the hardware is ok. 
I've also re-checked connections and tried with and without IRQ connected.
Then I've tried to restart from basics, writing a really simple software, still not working.
Also tried to work with fixed and variable payload: nothing
Maybe I'm working with some default that I should change? How?

This outputs with C++ program

STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0x4152443031 0x5250493031
RX_ADDR_P2-5     = 0xc3 0xc4 0xc5 0xc6
TX_ADDR          = 0x4152443031
RX_PW_P0-6       = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA            = 0x00
EN_RXADDR        = 0x03
RF_CH            = 0x4c
RF_SETUP         = 0x07
CONFIG           = 0x0e
DYNPD/FEATURE    = 0x00 0x00
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_MAX

And this from Javascript program
SPI device:      /dev/spidev0.0
CE GPIO:         24
IRQ GPIO:        25
STATUS:          0xe RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0â1:    0x4152443031 0x5250493031
RX_ADDR_P2â5:    0xc3 0xc4 0xc5 0xc6
TX_ADDR:         0x4152443031
RX_PW_P0â5:      0x0 0x0 0x0 0x0 0x0 0x0
EN_AA:           0x00
EN_RXADDR:       0x02
RF_CH:           0x4c
RF_SETUP:        0x03
CONFIG:          0x0e
DYNPD/FEATURE:   0x00 0x07
Data Rate:       1Mbps
Model:           nRF24L01+
CRC Length:      16 bits
PA Power:        PA_LOW

Please help me: I'm getting mad!
Thank you
Luciano

natevw, 2018-02-05:
Sorry, lots of people having trouble with this recently and I'm kind of swamped with other work to investigate now :-(

You can try set `radio._debug = true` might give you more clues. Only thing I notice from here is lower power from JS. Is your JS side supposed to receive, or transmit, or both?

luckysistemi, 2018-02-06:
Thank you for your reply.
the module should both transmit and receive. The power is not a problem for the moment, because the two modules are on the same table.

I finally managed to receive packets, adding some code that maybe cause problems to internal status of the driver:
`radio.begin(function() {
	radio.setStates({FEATURE:0x00});
	var rx = radio.openPipe('rx', 0x5250493031); // Listen at address
	var tx = radio.openPipe('tx', 0x4152443031); // Send to address
	radio.setStates({RX_PW_P0:0x20});
	radio.setStates({RX_PW_P1:0x20});
	radio.setStates({DYNPD:0x00});
...
`
With this code I receive packets (the content is reversed, but there are no packet loss)
Theese setStates are needed to make all the printDetail options identical to what I have with C++ program

The problem remains the trasmission with ack: no packet are transmitted. I think because no ack is received

this is the last fragment of debug:
`...
setStates { TX_ADDR: <Buffer 41 52 44 30 31>,
  PRIM_RX: false,
  RX_ADDR_P0: <Buffer 41 52 44 30 31> }
execCommand [ 'R_REGISTER', 0 ] 1
 - exec read: <Buffer 0e 0f>
execCommand [ 'W_REGISTER', 0 ] [ 14 ]
blocked for 130Âµs.
execCommand [ 'W_REGISTER', 10 ] <Buffer 41 52 44 30 31>
execCommand [ 'W_REGISTER', 16 ] <Buffer 41 52 44 30 31>
execCommand W_TX_PAYLOAD <Buffer 2b 2b 2b 2b 30 30 30 30 2b 2b 2b 2b 31 32 33>
`
last line is the result of tx.write. Then the program stop doing anything, no other output is printed.