natevw/node-nrf #64: I can't init connection with NRF

engmarconi, 2019-06-19:


engmarconi, 2019-06-19:
> this my nodejs code on Pi3
> const cePin = 24;
> const irqPin = 25;
> const spiDev = "/dev/spidev0.0";
> pipes = [0x746D610000, 0x0000616D74];
> 
> var nrf = require('nrf');
> var radio = nrf.connect(spiDev, cePin, irqPin);
> radio._debug = true
> 
> radio.channel(115);
> radio.dataRate('1Mbps');
> radio.crcBytes(2);
> radio.autoRetransmit({ count: 15, delay: 4000 });
> 
> radio.begin(function () {
> var rx = nrf.openPipe('rx', pipes[0]);
> var tx = nrf.openPipe('tx', pipes[1]);
> rx.on('data', function (d) {
> console.log("Got data, will respond", d);
> tx.write(d);
> });
> tx.on('error', function (e) {
> console.warn("Error sending reply.", e);
> });
> });
> 
> _But I got the following response_
> setStates { RF_CH: 115 }
> setStates { RF_DR_LOW: false, RF_DR_HIGH: false }
> setStates { EN_CRC: true, CRCO: 1 }
> setStates { ARC: 15, ARD: 15 }
> Set CE low.
> blocked for 130µs.
> Set CE low.
> blocked for 130µs.
> execCommand FLUSH_TX 0
> execCommand [ 'W_REGISTER', 5 ] [ 115 ]
> execCommand FLUSH_RX 0
> execCommand [ 'R_REGISTER', 6 ] 1
> setStates { PWR_UP: true,
> PRIM_RX: false,
> EN_RXADDR: 0,
> RX_DR: true,
> TX_DS: true,
> MAX_RT: true,
> EN_DPL: true,
> EN_ACK_PAY: true,
> EN_DYN_ACK: true }
> 
> * exec read: null
>   /home/pi/Documents/DreamHome/NRF_demo/Pi/node_modules/nrf/index.js:150
>   var val = d[0],
>   ^
> 
> TypeError: Cannot read property '0' of null

natevw, 2019-06-19:
Uh-oh, I did recently push an update to `pi-spi` for node v12 compatibility and perhaps that broke something :-/ I filed https://github.com/natevw/pi-spi/issues/31 to followup if that is the case.

If you are able to `npm install pi-spi@1.1.0` does the problem go away?

engmarconi, 2019-06-19:
Thanks @natevw for reply. I updated pi-spi and error not appeared again, but I changed some config like data rate and transmit power, then print details I got different values.
//Code
const cePin = 24;
const irqPin = 25;
const spiDev = "/dev/spidev0.0";
pipes = [0x7878787878, 0xB3B4B5B6F1];

var nrf = require('nrf');
var radio = nrf.connect(spiDev, cePin, irqPin);
//radio._debug = true

radio.channel(115).transmitPower('PA_MAX').dataRate('250kbps').crcBytes(2)
.autoRetransmit({count:15, delay:4000}).begin(function () {
  var rx = radio.openPipe('rx', pipes[0]);
  var tx = radio.openPipe('tx', pipes[1]);
  rx.on('data', function (d) {
    console.log("Got data, will respond", d);
    tx.write(d);
  });
  tx.on('error', function (e) {
    console.warn("Error sending reply.", e);
  });
  radio.printDetails();
});

//Details
SPI device:	 /dev/spidev0.0
CE GPIO:	 24
IRQ GPIO:	 25
STATUS:		 0x0 RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=0 TX_FULL=0
RX_ADDR_P0–1:	 0x0000000000 0x0000000000
RX_ADDR_P2–5:	 0x00 0x00 0x00 0x00
TX_ADDR:	 0x0000000000
RX_PW_P0–5:	 0x0 0x0 0x0 0x0 0x0 0x0
EN_AA:		 0x00
EN_RXADDR:	 0x00
RF_CH:		 0x0
RF_SETUP:	 0x00
CONFIG:		 0x00
DYNPD/FEATURE:	 0x00 0x00
Data Rate:	 1Mbps
Model:		 nRF24L01
CRC Length:	 Disabled
PA Power:	 PA_MIN

natevw, 2019-06-19:
@engmarconi Okay, sorry about the pi-spi issue — I will have to fix that and for now you should keep the older one installed. With that fixed, it looks like it is not actually talking to the NRF device (all data is 0) so please double-check your wiring and which pin connections you are using as a next step.

engmarconi, 2019-06-19:
Can you support me with standard connection with pi3?
Regards

natevw, 2019-06-20:
Matching the diagram at <https://pinout.xyz/pinout/spi> with your configuration:

```
const cePin = 24;
const irqPin = 25;
const spiDev = "/dev/spidev0.0";
```

The CE (RX/TX) pin of the transceiver should be on `BCM 24` [header pin 18] and the IRQ on `BCM 25` [header pin 22].

Then for spidev0.0 that means SPI0 bus using its CE0 pin:

* transceiver MOSI — `BCM 10` [header 19]
* transceiver MISO — `BCM 9` [header 21]
* transceiver SCK — `BCM 11` [header 23]
* transceiver CSN pin — `BCM 8` [header 24]

Hope this helps!

engmarconi, 2019-06-20:
Thanks @natevw, I got right config now.
Is there any pi to arduino example?

engmarconi, 2019-06-20:
According to this code:
//Code
const cePin = 24;
const irqPin = 25;
const spiDev = "/dev/spidev0.0";
pipes = [0x7878787878, 0xB3B4B5B6F1];

var nrf = require('nrf');
var radio = nrf.connect(spiDev, cePin, irqPin);
//radio._debug = true

radio.channel(115).transmitPower('PA_MAX').dataRate('250kbps').crcBytes(2)
.autoRetransmit({count:15, delay:4000}).begin(function () {
var rx = radio.openPipe('rx', pipes[0]);
var tx = radio.openPipe('tx', pipes[1]);
rx.on('data', function (d) {
console.log("Got data, will respond", d);
tx.write(d);
});
tx.on('error', function (e) {
console.warn("Error sending reply.", e);
});
radio.printDetails();

Is TX_ADDR right? I set tx pipe to pipes[1] = "0xB3B4B5B6F1"
[image]

natevw, 2019-06-21:
If the Arduino is what you show in second screenshot, with TX_ADDR of 0x7878787878 then yes `rx` on `pipes[0] = 0x7878787878` should be correct on Pi.

I do not have an Arduino example handy, but if I remember I tested it with RF24 library (https://github.com/maniacbug/RF24) maybe it was <https://github.com/nRF24/RF24/blob/master/examples/pingpair_ack/pingpair_ack.ino> file? I will have to look on an older laptop for more details.

engmarconi, 2019-06-22:
this screenshot for pi side  not Arduino

engmarconi, 2019-06-22:
NRF_node code on Pi3
===============
const cePin = 24;
const irqPin = 25;
const spiDev = "/dev/spidev0.0";
pipes = [0x544d52687C, 0xB3B4B5B6F6]

var nrf = require('nrf');
var radio = nrf.connect(spiDev, cePin, irqPin);
//radio._debug = true
var count = 0;

radio.channel(115).transmitPower('PA_MAX').dataRate('250kbps').begin(function () {
  var rx = radio.openPipe('rx', pipes[0], { autoAck: false });
  var tx = radio.openPipe('tx', pipes[1], { autoAck: false });
  rx.on('data', function (d) {
    console.log("Got data, will respond", d);
  });
  tx.on('ready', function () {
    console.log("TX ready.");
    setInterval(function () {
      tx.write(count + "");
      console.log("send", count);
      count++;
    }, 2000);
  })
  tx.on('error', function (e) {
    console.warn("Error sending reply.", e);
  });
  radio.printDetails();
});

-------------------------------------------------------------------
Arduino Uno code to ready data sent from pi
==========================
#include <SPI.h>
#include "nRF24L01.h"
#include "RF24.h"
#include "printf.h"

RF24 mRadio (9, 10);

const uint64_t pipes[1] = { 0xB3B4B5B6F6LL};
bool initDone = false;
byte pipeNum = 0;
char dataRecieve[] = "";

void setup() {
  Serial.begin(115200);
  mRadio.begin();
  mRadio.setChannel(115);
  mRadio.setPALevel(RF24_PA_MAX);
  mRadio.setDataRate(RF24_250KBPS);
  mRadio.openReadingPipe(0, pipes[0]);
  mRadio.startListening();
  Serial.println("Start listening");
}

void loop() {
  if (mRadio.available()) {
    while (mRadio.available(&pipeNum)) {
      mRadio.read( &dataRecieve, sizeof(dataRecieve) );
    }
    Serial.print("Recieve from : ");
    Serial.println(pipeNum);
    Serial.print("Package:");
    Serial.println(dataRecieve);
  }
}
----------------------------------------------------------------------
This pi output:
[image]

Although I changed pipes to [0x544d52687C, 0xB3B4B5B6F6] but still RX_ADDR_P0–1: 0x7878787878 0x544d52687c
I reset spi using these commands:
sudo modprobe -r spi_bcm2835
sudo modprobe spi_bcm2835

I can't receive on arduino, can you help me?
Regards
Marco

engmarconi, 2019-06-22:
reopen it

natevw, 2019-06-25:
Sorry, I tried to get a Pi set up again to test but ran into some trouble getting the OS/node updated and probably won't be able to get back to it until next Monday.