natevw/node-nrf #66: Cannot install nrf via npm

JockeyDoe, 2020-11-06:
```
239 verbose stack Error: epoll@0.1.22 install: `node-gyp rebuild`
239 verbose stack Exit status 1
239 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
239 verbose stack     at EventEmitter.emit (events.js:314:20)
239 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
239 verbose stack     at ChildProcess.emit (events.js:314:20)
239 verbose stack     at maybeClose (internal/child_process.js:1021:16)
239 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
240 verbose pkgid epoll@0.1.22
241 verbose cwd /home/pi/bma-central
242 verbose Linux 5.4.72-v7+
243 verbose argv "/usr/bin/node" "/usr/bin/npm" "install" "nrf"
244 verbose node v12.19.0
245 verbose npm  v6.14.8
246 error code ELIFECYCLE
247 error errno 1
248 error epoll@0.1.22 install: `node-gyp rebuild`
248 error Exit status 1
249 error Failed at the epoll@0.1.22 install script.
249 error This is probably not a problem with npm. There is likely additional logging output above.
250 verbose exit [ 1, true ]
```

natevw, 2020-11-27:
Hmm, it's probably well past time to test https://github.com/natevw/pi-pins with a more up-to-date epoll dependency. Hopefully there's not too many changes to deal with but iirc the usage there was fairly minimal anyway. Thanks for the report!

JockeyDoe, 2020-12-22:
Is there a possibility for me to fix it myself?

natevw, 2020-12-22:
Ah, sorry… yes it might be. So this library uses the `pi-pins` library (also one I wrote) to drive the GPIO pins, and that library depends on https://github.com/fivdi/epoll but the version I'm referencing there is super old. **But** a quick glance at that library looks like the calls I am making should still work.

So theoretically all that needs to be done is to update https://github.com/natevw/pi-pins/blob/8f984af/package.json#L24 to say `"^4.0.0"` instead of `"^0.1.3"` — and of course test to make sure that actually works as it used to. As a temporary workaround does it work to first `npm install epoll@4.0.0` followed by `npm install nrf`?

JockeyDoe, 2021-09-09:
Sorry for the late answer:
Nope, that doesn't work for me

`gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:400:28)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)
gyp ERR! System Linux 5.10.60-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /root/bma-central/node_modules/pi-pins/node_modules/epoll
gyp ERR! node -v v14.17.6
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! epoll@0.1.22 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the epoll@0.1.22 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-09-08T14_33_43_094Z-debug.log`