natevw/node-nrf #68: Error: EINVAL: invalid argument write

darienlabbe, 2024-03-28:
I am currently working on a personal project and I just intend on making a simple page that contains a checkbox that when selected sends a signal to my relay which allows power to flow and turn on a motor. I know that everything works I am just struggling to get node to work. I have installed the latest node and npm as well as onoff and socket.io for this as well. Below is the error that I am getting and I cannot for the life of me figure out what is wrong. If it helps I am using a Raspberry 4B with the arm64 latest os as of 3 days ago (from writing this). Anything would help!

```
node:fs:2342
    return binding.writeFileUtf8(
                   ^

Error: EINVAL: invalid argument, write
    at Object.writeFileSync (node:fs:2342:20)
    at exportGpio (/home/darien/Code/NodeJS/node_modules/onoff/onoff.js:18:8)
    at new Gpio (/home/darien/Code/NodeJS/node_modules/onoff/onoff.js:172:36)
    at Object.<anonymous> (/home/darien/Code/NodeJS/MotorControl/webserver.js:5:11)
    at Module._compile (node:internal/modules/cjs/loader:1368:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1426:10)
    at Module.load (node:internal/modules/cjs/loader:1205:32)
    at Module._load (node:internal/modules/cjs/loader:1021:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:142:12)
    at node:internal/main/run_main_module:28:49 {
  errno: -22,
  code: 'EINVAL',
  syscall: 'write'
}

Node.js v21.7.1
```

natevw, 2024-03-28:
It looks like you might be having trouble with https://www.npmjs.com/package/onoff though, and your problem wouldn't be related to this `node-nrf` library here. Maybe what https://github.com/fivdi/onoff/issues/197#issuecomment-1806247567 is saying about the way pins are labelled between the Pi and how `onoff` (/Linux itself) numbers them could be the trouble?

darienlabbe, 2024-03-30:
Thank you! Yeah it doesn't make a lot of sense to me but the GPIO pins have a very difficult identity than I was expecting. I was working with GPIO pin 24 which ended up being 536. Thanks for your help though!

closed 2024-03-30 by darienlabbe